photobooth-app/photobooth-app #450: add cropping/zoom

mgineer85, 2025-02-28:
That would be incredible if you could add the cropping feature to the backend. 
I've played around with v4l2-ctl on the CLI with no luck, so any help getting that up and running would be fantastic. Thanks as always for your help an incredible work!

_Originally posted by @sckowalski in https://github.com/photobooth-app/photobooth-app/discussions/449#discussioncomment-12356880_

mgineer85, 2025-03-02:
check if all backends support some kind of cropping or needs to implement separately in post (might be more inefficient)
- picamera2: https://github.com/raspberrypi/picamera2/blob/main/examples/zoom.py
- cv2: https://stackoverflow.com/a/57355537 (since it's in np anyways, can use simple matrix manipulation)
- gphoto2: to do
- digicamctrl: will not add as support will drop in future probably
- v4l /linuxpy:  todo

mgineer85, 2025-03-26:
So, to implement this in backend, cropping needs 3 algorithms for the backends variable input and expected output=jpeg:
- jpeg->jpeg (v4l2/gphoto/digicamctrl)
- numpy -> jpeg (cv2)
- picamera -> jpeg (native support)

maybe cv2 can be replaced by jpeg->jpeg also.
most efficient seems turbojpeg. if turbojpeg is not installed, fallback to numpy with additional decoding/encoding.

FUF1, 2026-02-17:
I look forward to that! 👁️ Doing some post production like cropping in the pipeline would be super helpful for custom things. As an example: I am having fun with an attached thermal imaging camera for a science education photobooth. It works. But since the camera only does a screencast via UVC, all the menu items are visible in the shot. It would be very nice to edit the image (in this case cropping, but maybe features like blurry edges, rounded corners, drop shadows, reflection and more would be nice too).

As an extendable option: Maybe there is a way to capture the image on disk with a specified variable as a name identifier, to very quick local image processing with kind of a "watch folder" script, and then be able to set the photobooth to ingest the image back in based on the variable name (with something like an "_edit" or so). This would make the setup killer-universal for all kinds of fun stuff.

By the way: I don't think the camera really releases a capture. Instead the photobooth seems to just make a "screenshot". This is a very nice default behaviour, since it allows to use all kinds of cameras. As a future feature having the option to ingest a network stream as a camera would be very nice to create a sophisticated external pipeline.

Just as a fan-sidenote: I just love the extendability of this photobooth-app with the GPIOs, WLED, API-calls and more. Just never saw any other photobooth solution just as suitable for tinkeres as this one! You did such a great job!