photobooth-app/photobooth-app #451: Stop live video stream if system is idle  / not using video feed

dtimes, 2025-03-01:
**Is your feature request related to a problem? Please describe.**
Camera is running all the time even if displaying the slideshow due to system being idle for an extended period of time, depending on the camera model this could cause stress to the image sensor (higher heat -> higher noise) + not all older DSLR capable of livestream are good in running extended times on live view . not sure about webcams but depending on model they heat up at least 

**Describe the solution you'd like**
live view / camera stream disabled if system is in idle mode (showing gallery ) , live stream turned back on once system is back to active state 

**Describe alternatives you've considered**
probably using a different camera which is mirrorless anyways / optimized for unlimited live view time , but I've got some older full frame canon SLRs on stock and I don´t want to buy new hardware for this purpose 

**Additional context**

As always thanks a lot for your work !

mgineer85, 2025-04-15:
So, with the upcoming release the frontend disconnects and reconnects if it needs the stream.
In future this could be used to close the camera after an additional timeout maybe.

dtimes, 2025-04-16:
thanks a lot for your work again, would be great to have the option to pause /disable the live view on the camera if it´s not in use ( shown on screen

mgineer85, 2025-04-16:
Not too simple because there are several backends, each with it's own implementation to be considered.
So whenever there is time... I would be happily accept non-chatgpt contributions :)

dtimes, 2025-06-23:
thanks , looks interesting,  can you add some details what this is actual doing ? can´t test at the moment and I don´t fully understand what this is doing and what exactly the finished event is.  maybe I am understanding the script wrong but the two curl requests at the end are sent without a delay which results in liveview disabled and directly enabled again ? (please correct me if I´m wrong), not sure how this helps cooling down the sensor.

foxylion, 2025-06-23:
Yes, it is disabling and enabling directly after afterwards. But the live stream is only started when required by starting a new photo sequence.

The finished event ist triggered after the photo sequence has completed.

dtimes, 2025-06-23:
ok , thanks for adding the details. If there is an event matching the start of the slideshow it might be a good point to trigger (not sure if there is one) , but I will try your script today or tomorrow.

dtimes, 2025-06-23:
tested and it works (kind of ) , this disables the liveview once a photo has been captured for the time it´s shown on the screen ( timer expired or manually skipped) and liveview is enabled again. for me this introduces a small (unwanted delay) from photo review to live stream again.   also this does not cover the box going from live view to the slideshow mode , if the box is idle but last screen was live view it will keep the live view active even if just the slideshow is running.   ( kind of back to the start of the question / I think there is no "box is idle" event yet)

mgineer85, 2025-06-24:
Hey, I tried to add it to the gphoto2 backend: https://github.com/photobooth-app/photobooth-app/commit/bdacf75ab4b42f9f875bb6de4ac2cca4d3ecfa60

the automated tests work, but I did not test it on real hardware.  If anybody is willing to try it I would appreciate.
In theory, the camera should disable after 20 seconds when no livestream is requested.
The relevant files to modify are gphoto2.py and abstractbackend.py.

dtimes, 2025-06-24:
ok thanks a lot , I can test this tomorrow

dtimes, 2025-06-25:
might be me applying the patch the wrong way ( only modified gphoto2.py and abstractbackend.py ) , gphoto2 backend  crashes:

2025-06-25 22:53:40,306 [    INFO] loaded backends: ['0:WebcamV4lBackend', '1:Gphoto2Backend'] (aquisition.py:56)
2025-06-25 22:53:40,480 [    INFO] start running service logic for ResilientService (resilientservice.py:50)
2025-06-25 22:53:40,521 [   DEBUG] WebcamV4lBackend starting as resilient service (resilientservice.py:88)
2025-06-25 22:53:40,521 [    INFO] start running service logic for ResilientService (resilientservice.py:50)
2025-06-25 22:53:40,521 [    INFO] trying to open camera index=self._config.device_identifier='0' (webcamv4l.py:197)
2025-06-25 22:53:40,521 [   DEBUG] Gphoto2Backend starting as resilient service (resilientservice.py:88)
2025-06-25 22:53:40,544 [    INFO] deleted 0 outdated items from the cache (collection.py:227)
2025-06-25 22:53:40,545 [    INFO] initialized DB, found 683 images (collection.py:269)
2025-06-25 22:53:40,575 [    INFO] starting shareservice worker_thread (qrshare.py:57)
2025-06-25 22:53:40,576 [   DEBUG] QrShareService started - it tries to connect to dl.php on regular basis now. (qrshare.py:41)
2025-06-25 22:53:40,578 [    INFO] FileTransferService started. (filtetransfer.py:37)
2025-06-25 22:53:40,579 [    INFO] services status: ['LoggingService: started', 'PluginManagerService: started', 'AquisitionService: started', 'MediacollectionService: started', 'InformationService: started', 'ProcessingService: started', 'SystemService: started', 'ShareService: started', 'GpioService: disabled', 'QrShareService: started', 'FileTransferService: started', 'ConfigurationService: started'] (container.py:63)
2025-06-25 22:53:40,579 [    INFO] started container (container.py:64)
2025-06-25 22:53:40,627 [    INFO] webcam device index: 0 (webcamv4l.py:199)
2025-06-25 22:53:40,628 [    INFO] webcam name: AnkerWork C310 Webcam (webcamv4l.py:200)
2025-06-25 22:53:40,629 [    INFO] switch_mode to lores requested (webcamv4l.py:116)
2025-06-25 22:53:40,633 [    INFO] requested lores-resolution is 1920x1080, format MJPG (webcamv4l.py:131)
2025-06-25 22:53:40,633 [    INFO]    actual lores-resolution is 1920x1080, format MJPEG (webcamv4l.py:132)
2025-06-25 22:53:40,658 [    INFO] lifecycle hook installing signal to handle app shutdown (application.py:46)
2025-06-25 22:53:40,685 [    INFO] Manufacturer: Canon.Inc
Model: Canon EOS 5D Mark IV
  Version: 3-1.3.3
  Serial Number: xxxxxxxxxxxxxxxxxxxxxxxxxx
Vendor Extension ID: 0xb (1.0)
Vendor Extension Description:

Capture Formats: JPEG
Display Formats: Association/Directory, Script, DPOF, MS AVI, MS Wave, JPEG, CRW, Unknown(b103), Unknown(bf02), Defined Type, Unknown(b104), Unknown(b105)

Device Capabilities:
        File Download, File Deletion, File Upload
        No Image Capture, No Open Capture, Canon EOS Capture, Canon EOS Capture 2
        Canon Wifi support

Storage Devices Summary:
store_00010001:
        StorageDescription: CF
        VolumeLabel:
        Storage Type: Removable RAM (memory card)
        Filesystemtype: Digital Camera Layout (DCIM)
        Access Capability: Read-Write
        Maximum Capability: 64014548992 (61049 MB)
        Free Space (Bytes): 46662156288 (44500 MB)
        Free Space (Images): -1

Device Property Summary:
Property 0xd402:(read only) (type=0xffff) 'Canon EOS 5D Mark IV'
Property 0xd407:(read only) (type=0x6) 1
Property 0xd406:(readwrite) (type=0xffff) 'Unknown Initiator'
Property 0xd303:(read only) (type=0x2) 1
Battery Level(0x5001):(read only) (type=0x2) Enumeration [100,0,75,0,50] value: 99% (99)
 (gphoto2.py:185)
2025-06-25 22:53:40,686 [   DEBUG] capturetarget value empty, ignore (gphoto2.py:162)
2025-06-25 22:53:40,686 [    INFO] start running service logic for ResilientService (resilientservice.py:50)
2025-06-25 22:53:40,699 [    INFO] successfully connected to shareservice dl.php script (qrshare.py:74)
2025-06-25 22:53:40,877 [    INFO] livestream starting on backend backend_to_stream_from=Gphoto2Backend (aquisition.py:125)
2025-06-25 22:53:40,887 [   DEBUG] configure camera optimized for idle/video (gphoto2.py:149)
2025-06-25 22:53:40,887 [   DEBUG] iso value empty, ignore (gphoto2.py:162)
2025-06-25 22:53:40,887 [   DEBUG] shutterspeed value empty, ignore (gphoto2.py:162)
2025-06-25 22:53:41,582 [   ERROR] 'NoneType' object has no attribute 'get_data_and_size' (resilientservice.py:35)
Traceback (most recent call last):
  File "/home/photo/.local/pipx/venvs/photobooth-app/lib/python3.11/site-packages/photobooth/utils/resilientservice.py", line 51, in _run
    self.run_service()
  File "/home/photo/.local/pipx/venvs/photobooth-app/lib/python3.11/site-packages/photobooth/services/backends/gphoto2.py", line 234, in run_service
    img_bytes = memoryview(capture.get_data_and_size()).tobytes()
                           ^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'get_data_and_size'
2025-06-25 22:53:41,584 [CRITICAL] backend crashed, error: 'NoneType' object has no attribute 'get_data_and_size' (resilientservice.py:36)
2025-06-25 22:53:41,750 [    INFO] trying to recover from crash (resilientservice.py:72)
2025-06-25 22:53:41,751 [ WARNING] service failed (attempt 1). Retrying in 2s... (resilientservice.py:81)

dtimes, 2025-06-27:
switched to dev using cmd in documentation ( pipx install --force --system-site-packages git+https://github.com/photobooth-app/photobooth-app.git@main ) , same gphoto2 crash . reverted back to stable and gphoto2 connected camera works again

mgineer85, 2025-06-27:
Yes, just gphoto update to 2.6.1 is good.

mgineer85, 2025-06-27:
I think pipx has a inject arg for this purpose

mgineer85, 2025-06-28:
Hey, pls wait before you investigate further.  I need to clarify on the correct usage of the lasted gphoto lib https://github.com/jim-easterbrook/python-gphoto2/discussions/191

dtimes, 2025-06-28:
ok thanks for the update  , I´ve already injected the new gphoto using  pipx inject photobooth-app gphoto2==2.6.1. With this photobooth-app starts normally and seems to work fine ( did not have much time for testing, will postpone until issue above is resolved)

mgineer85, 2025-07-21:
hi, yeah, i think the dev branch should be good to go for a test.

mgineer85, 2025-07-27:
Hey @dtimes, there is a release candidate for v8 released.

If you could test the latest version, I would appreciate.
https://pypi.org/project/photobooth-app/8.0.0rc1/

If you installed via pipx, uninstall the app and then install as follows.
`pipx install --system-site-packages photobooth-app==8.0.0rc1 --pip-args='--prefer-binary'`

Find more information about changes in the notes: https://github.com/photobooth-app/photobooth-app/releases/tag/v8.0.0rc1

I close the issue for now, if there is an issue, please post here.

closed 2025-07-27 by mgineer85