photobooth-app/photobooth-app #457: Frame Overlay / Settings

mgineer85, 2025-03-05:
_Originally posted by @PixelAlarm in https://github.com/photobooth-app/photobooth-app/discussions/453#discussioncomment-12400215_

mgineer85, 2025-03-05:
maybe use https://docs.pydantic.dev/latest/api/types/#pydantic.types.FilePath

mgineer85, 2025-03-05:
To simplify this, we might on first(/every?) start copy the demo-assets over to userdata.
This way there is no fallback handling needed and the file checks are easy. They just validate for real files instead of a file OR if it fails check for the availability in the demo assets data...

PixelAlarm, 2025-03-07:
> To simplify this, we might on first(/every?) start copy the demo-assets over to userdata. This way there is no fallback handling needed and the file checks are easy. They just validate for real files instead of a file OR if it fails check for the availability in the demo assets data...
> 
> Need to ensure that the demo assets is copied before any validation of the config takes place because then the app startup would fail.

I'm wondering where the actual demo frames are stored :)
On a fresh install, there is nothing in /userdata/frames, but the actual frame_image_photobooth-app.png is used somehow. (I'm not able to find the file, but it's displayed on the frontend).

It's smiliar to the plugins. They are active, but not shown in the path. I wanted to look into the WLED plugin, but can't find the folder.

mgineer85, 2025-03-07:
There is https://github.com/photobooth-app/photobooth-app/blob/main/src/photobooth/utils/helper.py#L34 get_user_files.
It checks if the file is avail in userdata and falls back to a directory in the installation.

The magic keeps the userdata directory clear but creates many issues and inconsistencies... For now I am testing a symlink that is automatically created below the userdata and redirects to the preinstalled assets.   With that you will have to live with one folder in the userdata that you cannot delete but thats life ;)

For the plugins there are two folders actually:
- built in is https://photobooth-app.org/reference/plugins/#plugins-built-in
- and to allow for easy tinkering, you have additionally the ./plugins folder scanned for files in the current working directory: https://photobooth-app.org/reference/plugins/#plugin-architecture

The idea is to get inspiration from the built-in ones and then place your files in CWD/plugins to tinker around...

closed 2025-03-09 by mgineer85

mgineer85, 2025-03-09:
Files Input Fields are now FilePath types and no more normal strings.

The input given will now be checked if the file is actually existing. If not, there will be a validation error.
Also there is a search function, start typing and it will list a filename that matches the input:

[image]

Striker62, 2025-03-18:
Hi, not shure if this is well known.
My frames located in /userdata/frames . Autocompletion only find the demoasset .

I'm on version 6.1

[image]

[image]

commit 3b854cd2 (2025-03-09): fix #457 by forcing chosen files to be existant.   also removed demoassets magic by creating a symlink during app startup to the demoassets that come with the app.