photobooth-app/photobooth-app #462: Disable Livestream between capture and result

PixelAlarm, 2025-03-07:
**Is your feature request related to a problem? Please describe.**
I'm using the gphoto2-backend for both live and stills.
On slower host hardware (Pi 4 4GB) with fast camera, there is an unpleasant optical occurrence:

My observation on a standard workflow:

1. Idle-Livestream
2. Countdown after button press
3. Takepic Msg Text / Capture
**4. Idle-Livestream again (while downloading picture?) for a short period**
5. Processing animation
6. Resultscreen

#4 Isn't nice in my opinion, since it just seems sluggish and not meant to be.

**Describe the solution you'd like**
A blank/black screen instant the processing animation while downloading would make it look much better.

**Describe alternatives you've considered**
Would also be nice to have some capture animation (aperture closing or similar) and then directly the processing animation or blank screen.

**Additional context**
I can try to take a video, if this is helpful.

mgineer85, 2025-03-07:
No need for a video. I hate this very much also , but not so much to change it yet 😅

Will change it in future somehow.

mgineer85, 2025-03-07:
Yeah, it's just how the job processor states are communicated to the frontend and between the changes this can happen...

mgineer85, 2025-03-26:
This one will be fixed in next release 🤌🏻

commit 2037ba3e (2025-03-24): major rewrite of processingmachine. mainly to improve maintenance by separating classes and improve linting.

also fixes #462
closed 2025-03-26 by mgineer85