photobooth-app/photobooth-app #473: Posting this as an idea, since I don't know if it's possible and useful to others.

mgineer85, 2025-03-14:
Posting this as an idea, since I don't know if it's possible and useful to others.

**Is your idea request related to a problem? Please describe.**
The live view from gphoto2 is working pretty good already, but there is some room for improvement.
I have a DSLM that is able to output clean HDMI.
The video is really smooth when captured via USB capture card and used with v4l2.

The setup I have in mind (and tested already) is using two backends:
- gphoto2 for stills
- webcamv4l2 for video that is feeded by HDMI capture from same camera (plug and play uvc HDMI capture card)

The problem with this setup:
- Clean HDMI only works in movie mode and not in PSAM modes. In other modes there are some overlays like AF frames. This might be ok for live view but not for captured video :)
- image capture with gphoto2 only works in PSAM modes

**Describe the solution you'd like**
The movie mode of my camera can be toggled with "gphoto2 --set-config eosmoviemode=0/1" though.
It would be great, if I could set my camera to moviemode after capture and switch to PSAM before capture.

Since this is only suitable for canon cameras, the prompt for other brands have to be researched.
Maybe it could be implemented as a generic field, where config-item and parameter can be filled in.

**Describe alternatives you've considered**
I tried setting the eosmoviemode using the commander before and after capture, utilizing gphoto2 prompts. But the device is blocked by python.

_Originally posted by @PixelAlarm in https://github.com/photobooth-app/photobooth-app/discussions/472_

closed 2025-03-26 by mgineer85

PixelAlarm, 2025-04-15:
Hi, I hat time to test this feature today. The switching of modes is working perfectly, but I think the current implementation has only limited use.

When using only gphoto2 backend for stills and video
- camera is switched to the right mode for the action and will stay in this mode until a different action is triggered.
- after an image action it will stay in the still mode
- after an video action it will stay in the eosmoviemode

--> The gphoto2 video stream seems to be the same for PASM and eosmoviemode, so maybe not that big of an advantage.

When gphoto2 is used for stills and PyAV is used for video:
- Camera is not switched to video mode for any action
- the eosmoviemode switch for the gphoto2 backend seems to have no effect

So I would propose following implementation, when the eosmoviemode switch is ON
- make use of eosmoviemode, even if another video backend is activated
- use eosmoviemode for all video related features (idle video, preview, recording)
- switch to stills mode only for taking the picture
- switch back to eosmoviemode when going to video function again

Thanks again for considering the initial request! 

Best regards

PixelAlarm, 2025-04-16:
I retested with the new beta 7.1.0b2

Configured 2 Backends:
1. gphoto2
2. pyAV

Stills set to Backend 0, Video set to Backend 1.
All video functions are served by hdmi grabber from camera hdmi via pyAV.

Camera set to M mode, the camera will never go to eosmoviemode, because pyAV is used for video and eosmoviemode is not respected.

I think eosmoviemode is only being used, when gphoto2 is used as video Backend.

Am I missing something maybe?

mgineer85, 2025-04-16:
Could you also provide the logs so I can see if the switching is applied correctly?

PixelAlarm, 2025-04-16:
> Could you also provide the logs so I can see if the switching is applied correctly?

Sure, here you go.
[log.txt](https://github.com/user-attachments/files/19775411/log.txt)

mgineer85, 2025-04-16:
maybe above commit helps for the busy issue, if not, I doubt we can handle it

PixelAlarm, 2025-04-17:
I‘m a bit lost, since the camera is one of the most recent ones and shutter 1/160, 1/200 always worked (my standard shutter speeds) and now all these values won’t work anymore.
The camera definitely supports 1/100 1/160 1/200 1/320. Nothing works.

I played with the config, turning eosmoviemode and viewfinder on and off, but always same output: Bad Parameters

Setting ISO to Auto, 100, 1600 works normally.

PixelAlarm, 2025-04-17:
I think it’s a typo in gphoto2.py

Should be „shutterspeed“ not „shutter_speed“ :)