photobooth-app/photobooth-app #474: CSS snippet mirror only stream

mgineer85, 2025-03-17:
Just for the records, this is the CSS snippet i put in `userdata/private.css`:

```
img#overlay-image {
    visibility: hidden;
}

#overlay-wrapper {
  background-image: url("/api/aquisition/stream.mjpg");
  background-size: cover;
  background-position: center;
  background-repeat: no-repeat;
}
```

Unfortunately one needs to add the path to the stream again, because the <img> tag cannot be used (the background and the image cannot be hidden separately from each other).

_Originally posted by @lumbric in https://github.com/photobooth-app/photobooth-app/discussions/453#discussioncomment-12525339_

lumbric, 2025-03-18:
**What this does:**
it hides the frame overlay during the countdown and re-adds the live stream as background to the parent div container. That means it removes the frame overlay during countdown. ATM one can only remove the frame during live stream before the countdown or during countdown and for the final image. This helps if you want it only for the final image, but never during the live stream and countdown period. So if you have text in the frame, it will never displayed mirrored.

mgineer85, 2025-04-04:
in current dev (and next release) this is not needed any more. the stream and the frame can be flipped independently.

closed 2025-04-04 by mgineer85