photobooth-app/photobooth-app #475: Filename auto Complet

Striker62, 2025-03-18:
I did not see that the issue 457 is closed. I added my coment there.

https://github.com/photobooth-app/photobooth-app/issues/457

My frames located in userdata/frames but the auto completion only show the demoasset.

I'm on 6.1

[image]

[image]

mgineer85, 2025-03-19:
Hi, it is correct as is.

You should to use
`WORKDIR/userdata` as custom folder for your files, instead you use
`WORKDIR/photobooth/userdata` to put your files.

That is why you see only demoassets, because the actual data is loaded from the other folder.

In theory you could still continue to use it this way, but it creates issues if you do not use the correct folder because you cant access them via http then.

Striker62, 2025-03-19:
Ok I See . Thanks

closed 2025-03-19 by Striker62