photobooth-app/photobooth-app #476: Onscreen keyboard not working in Photobooth app, outside Photobooth app it is working

WebfinityDesign, 2025-03-19:
**Describe the bug**
I like to use the onscreen keyboard to enter email addresses when sending the pictures to email, so I installed:
sudo apt install squeekboard wfplug-squeek

It is not working when I want to enter an email address inside Photobooth app, it does not pop up.
When I close Photobooth app and the browser and open a terminal and start typing the keyboard pops up.

**To Reproduce**
Steps to reproduce the behavior:
1. sudo apt install squeekboard wfplug-squeek
2. Make a picture and send to email
3. Try to enter emailaddress. No keyboard pops up
4. Close photobooth app and browser
5. Open a terminal and try to type. Keyboard pops up

**Expected behavior**
In photobooth app try to enter an email adress. Keyboard pops up.

**System (please complete the following information):**
 - OS: Bookworm 64bit
 - Browser: chrome

PixelAlarm, 2025-03-19:
Hi,
I think this is not a problem with photobooth-app, but with the virtual input of chromium.
I had some problems in the past with that.

Is this working, when you open up an empty browser tab and try to put in a URL?

There might be a workaround with setting some flags when starting chromium.

WebfinityDesign, 2025-03-20:
@PixelAlarm You are right. When I open a browser and try po enter a URL it doesn't popup eihter.
I also tried the same in Firefox. In Firefox the keyboard pops up when I want to enter an URL.
Do you know the workaround for Chrome or would it be better to switch to Firefox?

PixelAlarm, 2025-03-20:
> [@PixelAlarm](https://github.com/PixelAlarm) You are right. When I open a browser and try to enter a URL it doesn't popup either. I also tried the same in Firefox. In Firefox the keyboard pops up when I want to enter an URL. Do you know the workaround for Chrome or would it be better to switch to Firefox?

Try to edit the Exec command like so (inside the .desktop file)
`Exec=chromium --kiosk --fullscreen http://localhost:8000 --ozone-platform=wayland --enable-wayland-ime --wayland-text-input-version=3`

WebfinityDesign, 2025-03-20:
> Try to edit the Exec command like so (inside the .desktop file) `Exec=chromium --kiosk --fullscreen http://localhost:8000 --ozone-platform=wayland --enable-wayland-ime --wayland-text-input-version=3`

I tried the Exec command, but with that it still doesn't work.
I also tried to set Firefox as the default browser, but in the photobooth app I do have the same problem.
I'll try to figure out if there is a setting that works.

mgineer85, 2025-03-24:
Closing this as not related to the app itself. 
If anybody has working and approved command line args feel free to add these here so they can be added to the docs.

thx!

closed 2025-03-24 by mgineer85

bbo76, 2026-01-17:
Any workaround for this bug ? Impossible for me to show the virtual keyboard on admin PIN input.

mgineer85, 2026-01-17:
Seems still to be an issue that is not solved for Kiosk Applications https://forums.raspberrypi.com/viewtopic.php?t=390053