photobooth-app/photobooth-app #478: Live stream lost after taking picture

WebfinityDesign, 2025-03-20:
**Describe the bug**
Not sure if this is a bug or a new setting in V6 that I'm missing.
When I start the photobooth I do have a livestream, but after taking the first picture I'm left with a stiil.
The still is not the same as the picture taken.

**To Reproduce**
Steps to reproduce the behavior:
1. Start up photobooth
2. There should be a livestream
3. Take a single picture
4. There is no more livestream

**Expected behavior**
Get back to livestream after taking a picture.

**System (please complete the following information):**
 - OS: Raspberry Pi Bookworm 64bit
 - Browser: chrome
 - Camera: Arducam Hawkeye 64MP

**Additional context**
Didn't have this problem in V4. I don't know about V5 because I switched to V6 before I had V5 working.

mgineer85, 2025-03-20:
I saw this once after the dependencies for the backend were updated. But never really again. I could think there might have been changes to uvcorn or fastapi.
The streaming code itself is not touched for some time…

mgineer85, 2025-03-22:
For whatever reason the camera stops deliver frames.
maybe you can see more errors in system logs or if you start the app directly on the console on the stdout live log?

Following to getting no images from the camera any more, the camera is restarted. The browser does not automatically restart the stream though. When the stream freezes and you refresh the browser is should work again.

```
2025-03-22 18:56:46,180 [   ERROR] camera stopped delivering frames, error  (picamera2.py:399)
2025-03-22 18:56:46,308 [    INFO] device signals is ready to deliver now=False (abstractbackend.py:191)
2025-03-22 18:56:46,312 [    INFO] _generate_images_fun left (picamera2.py:424)
2025-03-22 18:56:46,407 [    INFO] starting devices... (aquisition.py:110)
2025-03-22 18:56:46,412 [   DEBUG] photobooth.services.backends.picamera2 stop called (abstractbackend.py:171)
2025-03-22 18:56:46,415 [    INFO] device signals is ready to deliver now=False (abstractbackend.py:191)
2025-03-22 18:56:46,417 [    INFO] no picamera2 video encoder active that could be stopped (picamera2.py:284)
2025-03-22 18:56:46,628 [    INFO] Camera stopped (picamera2.py:1224)
2025-03-22 18:56:46,679 [   DEBUG] device not alive any more, stopping early lores image delivery. (abstractbackend.py:309)
2025-03-22 18:56:46,984 [    INFO] Camera closed successfully. (picamera2.py:641)
2025-03-22 18:56:46,986 [   DEBUG] stopping encoder (picamera2.py:163)
2025-03-22 18:56:46,987 [   DEBUG] photobooth.services.backends.picamera2 stopped (picamera2.py:165)
2025-03-22 18:56:46,987 [    INFO] starting device (aquisition.py:78)
2025-03-22 18:56:46,988 [   DEBUG] photobooth.services.backends.picamera2 start called (abstractbackend.py:161)
2025-03-22 18:56:46,989 [    INFO] device signals is ready to deliver now=False (abstractbackend.py:191)
2025-03-22 18:56:46,991 [    INFO] closing camera before starting to ensure it's available (picamera2.py:79)
2025-03-22 18:56:47,538 [    INFO] Initialization successful. (picamera2.py:458)
2025-03-22 18:56:47,550 [    INFO] Camera now open. (picamera2.py:480)
2025-03-22 18:56:47,598 [    INFO] Camera configuration has been adjusted! (picamera2.py:1068)
2025-03-22 18:56:47,610 [    INFO] Configuration successful! (picamera2.py:1073)
```

To investigate:
- why is no temp error img substitute sent to the browser as stream
- why the camera delivers no frames
- is it possible to automatically resume the frame after the cam is available again

I will check on my system here too. Maybe there is a hardware issue with the camera/connector?

WebfinityDesign, 2025-03-22:
I think we can close this issue. The reason seemed to be that the photobooth used an IP address in the netwrk that was already in use. After changing this it looks like the problem is solved. I didn't expect a double IP address could stop the camera stream.
If it happens again with the new IP address I will reopen this thread again

mgineer85, 2025-03-23:
I’ll keep it to addresse the other tasks at one point and close it later

mgineer85, 2025-03-25:
There is the webcodecs api nowadays. 
Maybe we can add a second endpoint to move on to more efficient H264/H265 streaming instead spitting out jpg frames.

Sample page, need to see how well it works on the pi boards.
https://w3c.github.io/webcodecs/samples/

mgineer85, 2025-03-26:
closing this continue in #482

closed 2025-03-26 by mgineer85

mgineer85, 2025-03-29:
So, I am experimenting streaming the preview via websockets.

Seems to work quite well, still need to sort out some edge cases.
Also there are separate options to mirror the preview and/or the overlaid frame.

https://github.com/user-attachments/assets/31f95611-5608-43e7-bd6e-07f479466a04