photobooth-app/photobooth-app #480: Security issue: dl.php makes all pictures public even if file identifier is not known

lumbric, 2025-03-26:
The share service allows to share pictures via a web server using extras/shareservice/dl.php. Pictures are uploaded as soon as the URL to the picture is accessed via the QR code. The 32 character long file identifier, which is part of the download link, suggests that the file is only accessible if the URL or the file identifier is known to the user.

However, there is an option to list all file identifiers, which allows to simply download all pictures.

**To Reproduce**
Steps to reproduce the behavior:
1. https://your-domain/dl.php?action=list
2. use one of the file identifiers, e.g. `[file_identifier] => e50504c9-73e9-484d-81c3-6a3d0084bb87`
3. download the file via  https://your-domain/dl.php?action=download&id=THE_IDENTIFIER

**Expected behavior**
I would expect the pictures to be either completely public, i.e. via a public list of a thumbnail gallery on the start page or to be truly hidden without knowing the precise URL.

**Solution**
The actions "list" and "upload_queue" should be protected using the $API_KEY similar to upload.
Does the file identifier contain (enough) random characters or can it be guessed easily?

mgineer85, 2025-03-26:
Makes sense to protect the routes, totally agree.

mgineer85, 2025-04-04:
The 'list' action is removed as it was used for debugging only anyways.

The upload_queue now checks the api key also. 

thanks for reporting!

commit 2892127a (2025-04-04): fix #480
closed 2025-04-04 by mgineer85