photobooth-app/photobooth-app #483: startup sequence

mgineer85, 2025-03-26:
not sure if it´s a chrome feature, but I´m starting it using xfce autostart and chrome starts before photobooth-app is ready (shows page not reachable first ) but automatically reloads page after a few seconds and shows the webui   ( I´ve added a delay of 20 seconds to the startup script of chrome anyways to avoid confusion ). I need the refresh button because sometimes the stream is stuck but works again after refreshing (mainly if DSLR fails to focus and no picture is taken, setup is with webcam for video preview feed and DSL for taking actual picture)

_Originally posted by @dtimes in https://github.com/photobooth-app/photobooth-app/discussions/479#discussioncomment-12626559_

mgineer85, 2025-03-26:
Maybe changing startup method to this works https://unix.stackexchange.com/questions/575527/opening-web-browser-on-boot
The app and browser can be set to start one after another using this method

WebfinityDesign, 2025-03-27:
If that down't work you can load a local file (logo, an image, etc.) in the browser at the browser startup to avoid the unreachable message. That was my initial idea to solve this unreachable message, but I still have to try it out.

mgineer85, 2025-03-31:
Maybe like this https://unix.stackexchange.com/a/519585
Plus another wantedby, requiredby PhotoBooth service…

mgineer85, 2025-04-04:
the service way seems not to work flawlessly...

also after starting the photobooth-app there might be some delay until the server is actually listening.

So best solution is probably to delay the start of chrome until the server is up.

`
until curl -s http://localhost:8000 >/dev/null; do sleep 1; done; chromium-browser --kiosk --disable-features=Translate --noerrdialogs --disable-infobars --no-first-run --ozone-platform=wayland --enable-features=OverlayScrollbar --start-maximized http://localhost:8000/
`
This is how it could be done.  It's added to the docs and works well on the latest Pi. 
https://photobooth-app.org/setup/installation/#labwc

closed 2025-04-04 by mgineer85