photobooth-app/photobooth-app #484: action related frame enabled

mgineer85, 2025-03-31:
Hi, I just checked again and I have no action related frame enabled:
[image]

And also disabled livestream frame overlay:
[image]

Live View Idle:
[image]

Live View Active (On Countdown):
[image]

_Originally posted by @PixelAlarm in https://github.com/photobooth-app/photobooth-app/discussions/453#discussioncomment-12548483_

mgineer85, 2025-04-02:
This is fixed in latest ui and next release.
the UI setting is ONLY active during idle mode, the action setting ONLY during countdown.
That means, if only UI is configured, during countdown the action needs to be configured also if there shall be an overlay.

closed 2025-04-02 by mgineer85

PixelAlarm, 2025-04-02:
Looking forward to it :)

So will there be an "Enable Livestream Overlay" in the action setting?
Or will the frame from be action always be shown on livestream?

mgineer85, 2025-04-02:
The one in the action will always be shown if defined.