photobooth-app/photobooth-app #491: PyAv set framerate on Mac

mgineer85, 2025-04-06:
Had time to dig into the problem just now.

I think the problem is, that the ffmpeg default framerate is 29.xxx and the macOS build-in devices only offer 30.0000 and 15.000 fps.

Using "ffmpeg -f avfoundation -framerate 30 -i "0" output.mkv" it works.
Using "ffmpeg -f avfoundation -i "0" output.mkv" it gives errors.

Maybe the framerate can be added for Darwin OS.

_Originally posted by @PixelAlarm in https://github.com/photobooth-app/photobooth-app/discussions/486#discussioncomment-12740431_

mgineer85, 2025-04-06:
Propose to add one or two settings to pyav backend:
`set_framerate:bool` default `False`
`framerate:int` default `30`

This way, Linux and Windows users are ready to go with the default settings. For Mac you probably need to turn on the enable switch and be fine.

If it still fails to open the camera you have the framerate available in the config to fiddle around.

closed 2025-04-06 by mgineer85

PixelAlarm, 2025-04-15:
I tested the latest version with my MacBooks internal Camera.
Initializations comes further than before, but still not able to use this:

- Devices Identifiers are not autocompleted
- When manually inserted rightly configured (using only exact name of the device), webcam indicator will constantly turn on and off
- It manages to get some device information like "frame format: <av.VideoFormat uyvy422, 1920x1080>" and throws error "Temporality not available" while trying to initialize

`2025-04-15 16:49:13,534 [   ERROR] [Errno 35] Resource temporarily unavailable: 'FaceTime HD-Kamera' (resilientservice.py:38)
Traceback (most recent call last):
  File "/photobooth-app/src/photobooth/services/backends/resilientservice.py", line 81, in _run
    self.run_service()
    ~~~~~~~~~~~~~~~~^^
  File "/photobooth-app/src/photobooth/services/backends/webcampyav.py", line 154, in run_service
    for frame in input_device.decode(input_stream):
                 ~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^
  File "av/container/input.pyx", line 215, in decode
  File "av/container/input.pyx", line 174, in demux
  File "av/container/core.pyx", line 303, in av.container.core.Container.err_check
  File "av/error.pyx", line 424, in av.error.err_check`

Maybe this helps a bit for further investigation.

mgineer85, 2025-04-15:
- device identifier is expected not to work because I have no possiblity to test the command to enumerate the devices. if you can confirm a command to enumerate, I could add it. Maybe it is similar to the linux command? https://github.com/photobooth-app/photobooth-app/blob/c14da0f5c5acbf879834e9091d3c85da45b82d1f/src/photobooth/utils/enumerate.py#L27
- I _think_ for avfoundation you put the index, not the name: https://ffmpeg.org/ffmpeg-devices.html#avfoundation  did you try a 0 or a 1?
- I have no idea what is going on. You have a complete logfile?

PixelAlarm, 2025-04-15:
I will try to look into the code and adapt something for macOS to get identifier working.

I think I can provide a logfile later.

commit a435c51a (2025-04-06): fix #491