photobooth-app/photobooth-app #493: Missing live video stream after saving the config in admin menu

msmedien, 2025-04-08:
**Describe the bug**
If I open the administration menu, change settings (regardless of the category), and then click Save Settings at the bottom right, the live video stream disappears when I exit the administration menu. I then have to restart the Systemctl service. (Sometimes even the entire system.) If I simply open the administration menu and close it without changing/saving any settings, the live video stream works without any problems.

**To Reproduce**
Steps to reproduce the behavior:
1. Open Admin Menu
2. Go to Config-Tab
3. Change any Options and click save
4. Leave the Admin Menu
5. The livestream is not visible anymore

**Expected behavior**
If I open the admin menu, change settings, and save them, the livestream should still be visible on the frontpage. This has always been the case in other versions of the photobooth-app.

**System (please complete the following information):**
 - OS: Raspberry Pi Bookworm 64bit - clean installation with all updates
 - Browser: Chromium
 - Version: 134.0.6
 - Installation via PipX
 - Camera for Live Stream: PiCamera v3 with all default settings in the photobooth-app 

**Additional context**
Had the problem in Photobooth-App 6.1.0 and now in 7.0.1. In other versions i had no problems.

mgineer85, 2025-04-08:
That’s because the backend is shutdown and might not restart properly. 
It’s not a critical issue, but I would like to solve it.
Can you provide logs from the action so I can see what happens?

msmedien, 2025-04-08:
That's the Log-File from today. I tried also a camera via gPhoto2 (Sony Alpha 6000). I think the start (system informations) and the end (changing back to the picamera v3 and disable the auto exposure function) of this Log-File is only important.  

[photobooth_20250408.log](https://github.com/user-attachments/files/19650017/photobooth_20250408.log)

I hope this will help a little bit. If you need more or something else, please let me know. :-)

mgineer85, 2025-04-08:
So after the config was saved, the backends restarted, then later the stream is requested. 

```
2025-04-08 13:38:38,774 [   DEBUG] capturetarget value empty, ignore (gphoto2.py:193)
2025-04-08 13:38:38,775 [    INFO] device signals is ready to deliver now=True (abstractbackend.py:176)
2025-04-08 13:38:38,775 [   DEBUG] photobooth.services.backends.gphoto2 started (gphoto2.py:93)
2025-04-08 13:38:38,776 [    INFO] device started (aquisition.py:91)
2025-04-08 13:38:41,880 [    INFO] ('127.0.0.1', 54538) - "WebSocket /api/aquisition/stream" [accepted] (websockets_impl.py:272)
2025-04-08 13:38:41,886 [    INFO] connection open (server.py:643)
2025-04-08 13:38:41,886 [    INFO] livestream started on backend backend_to_stream_from=<class 'photobooth.services.backends.gphoto2.Gphoto2Backend'> (aquisition.py:276)
2025-04-08 13:38:41,946 [    INFO] (gp_libusb1_read [libusb1.c:599]) 'libusb_bulk_transfer (port->pl->dh, port->settings.usb.inep, (unsigned char*)bytes, size, &curread, port->timeout)' failed: Input/Output error (-1) (port_log.py:20)
2025-04-08 13:38:41,947 [    INFO] (gp_port_read [gphoto2-port.c:427]) Reading 1024 = 0x400 bytes from port failed: Fehler beim Lesen des Ports (-34) (port_log.py:20)
```

When not reloading the service while saving the config, the DSLR can deliver a livestream normally without issues?
The gphoto2 library does not look to connect reliable to the camera. Maybe it's not a good combo?

For the picamera2, I see the error, I had it earlier - it seems to mess up internally in picamera2. Not sure if I can work around that, I will investigate.

So probably it's better if you just do not use the automatic reload functionality and restart the service manually (or via the dashboard). This way the complete app is restarted, not just the services.

mgineer85, 2025-04-16:
This should be fixed in the latest 7.1 beta now.

commit 822625eb (2025-04-08): maybe fix #493
commit 79c72575 (2025-04-08): Revert "maybe fix #493"

This reverts commit 822625ebf40c8cf4acd96e44962c7dbfe930b221.
PR #497 (2025-04-13): Refactor startstop
fixes #493 and #495
closed 2025-04-17 by mgineer85