photobooth-app/photobooth-app #495: Stream Recovery is not working gphoto2

Peda1996, 2025-04-08:
When the USB connection is interrupted (e.g. camera unplugged) and then reconnected after ~30 seconds, the livestream does not resume automatically.

Steps to Reproduce:
- Start the booth with configured gphoto2
- Remove the power from the the camera.
- Wait approximately 30 seconds.
- Turn the Camera back on.
- Observe that the livestream does not recover.
- Reloading leads to a white Background screen!.

Taking Photos is now also not working anymore.

mgineer85, 2025-04-08:
Please post logs of what is going on. I think its the backend not recovering from the faulty state, not the reconnection method.

mgineer85, 2025-04-08:
Pls disable the automatic reloading on config save. It seems to be buggy with some backends currently.

Peda1996, 2025-04-08:
That's the switch which is located above Save in the Admin Panel or? I disabled this switch, but it didn't seem to solve the issue.

mgineer85, 2025-04-09:
After it was once enabled in the session, it seems to be unstable. So did you once do a full restart after having had it enabled?
Checking logs later again, was on mobile yesterday.

mgineer85, 2025-04-09:
> That's the switch which is located above Save in the Admin Panel or? I disabled this switch, but it didn't seem to solve the issue.

Please check again. In the logs (might be partial only), the reloading was enabled all the time.
`2025-04-08 18:01:56,142 [    INFO] 127.0.0.1:42612 - "PATCH /api/admin/config/app?reload=true HTTP/1.1" 200 (httptools_impl.py:476)`

mgineer85, 2025-04-09:
Nevertheless, I am able to reproduce the stream is not recovering for gphoto2 after it failed once.

mgineer85, 2025-04-15:
just release v7.1.0b2 - I tested with my camera (Canon 1100D) and it resumes the stream if the http connection was broken as well as it's resuming now the camera connection properly.

Peda1996, 2025-04-16:
It works, thanks for your effort