photobooth-app/photobooth-app #499: Webcam format YUYV supported

mgineer85, 2025-04-13:
### Discussed in https://github.com/photobooth-app/photobooth-app/discussions/498

<div type='discussions-op-text'>

<sup>Originally posted by **naufilmetkar** April 13, 2025</sup>
Hello Sir,

I am trying to use a webcam which supports YUYV format and i am able to access it on my system using ffplay /dev/video0
after integration with the photobooth app it runs into an error. Run time error to be more precise. 

Is this format not supported or is there a way to work around this. Any help, suggestions will be much appreciated.

Regards,
Naufil Metkar</div>

naufilmetkar, 2025-04-14:
Thank you for your quick updates, this doesn't seem to work for me, even the MJPEG webcam doesn't work for me.

[photobooth_20250414.log](https://github.com/user-attachments/files/19729342/photobooth_20250414.log)

mgineer85, 2025-04-14:
It doesn't work because it's missing the turbojpeg system libs, have you installed the prerequisites?
https://photobooth-app.org/setup/installation/#install-system-dependencies-rpi

That command is tested on debian/Raspberry. If the libs are named different, you need to look up the correct packages.

mgineer85, 2025-04-15:
closing this one, if you encounter any issues, please open another issue. thank you!

by the way, which virtual camera software did you test/use?   Maybe its helpful to add to the documentation...

closed 2025-04-15 by mgineer85

naufilmetkar, 2025-04-16:
> closing this one, if you encounter any issues, please open another issue. thank you!
> 
> by the way, which virtual camera software did you test/use? Maybe its helpful to add to the documentation...

I used DroidCam with OBS Studio plugin which allows us to create a virtual webcam.
A good approach for people using android tablets as screen and it's camera for live streaming. Sadly, on iOS apps are not permitted to run in the background, so virtual camera and browser cannot be used simultaneously. 

Should be good to use a spare iphone as a high quality camera for stills and live streaming. The free version has no restrictions on the run time. Auto Focus function is quite good for pictures.

mgineer85, 2025-04-16:
👍 thank you, I will check where to add these information