photobooth-app/photobooth-app #500: Access to original camera media (gphoto2) missing in new version

Peda1996, 2025-04-14:
In previous versions, the original camera media (captured via gphoto2) used to be accessible under the media folder located within /home. However, in the current version this location seems to be missing. I could only partially restore a few files from /tmp after my last party, but most original files are lost.

Would be great to:
- Know the new path where original gphoto2 media is stored (if it still exists).
- Have a configuration option to explicitly persist original media in a specified folder.

mgineer85, 2025-04-15:
Hi - the media folder is considered to be an internal folder where files are stored for processing (and reprocessing if a filter is newly applied by the user).

All images from the camera are kept in the tmp folder. They remain there until manually deleted. Is this what happened?

Peda1996, 2025-04-15:
Hey yes, my problem was that somehow some files were missing from the tmp folder, so I assumed that they might have been deleted or smth similar. (eg had 120 processed vs 90 files within tmp and some files were clearly missing) 

Maybe they were overridden by another file with the same name, idk?

Also another reason for why I loved that images were located within a subfolder in media was that the copy to usb stick feature then also copied the original files directly to the usb. And the original files have the advantage of holding the best quality.

mgineer85, 2025-04-15:
You have collages enabled? Each collage generates +1 extra image next to the actual captures. Also true for GIF animation. That would explain the deviation.

I would like to keep the media folder clean and just put there what is used for processing, but yeah, maybe we can add another path for the actual originals again.

Either it's put in the job processor or in the abstractbackend so it is reusable for all backends.

Peda1996, 2025-04-15:
Well, I think the problem might be that I used the camera without SD card, therefore the camera just reused old names after some time. Simplest fix for this to not happen again would be to just append a timestamp or a uid to the filename or just to check whether the file already exists and append (2-x etc) accordingly...

Storing originals could also just be an toggleable option? But I don't know if this might clutter the interface too much.

mgineer85, 2025-04-15:
Hey, I think this simple refactoring should do it: bbfce819fb50e5ebf6d6cc8a6325531b7d516298 and replaces #501 also.

mgineer85, 2025-04-15:
just tested it, works :)

Peda1996, 2025-04-15:
Only downside to this would be that the filenames would be somewhat nonsensical, so not ideal for copying these then to an usb. What do you think of gphoto2_timestamp + (1-x).jpg instead?

mgineer85, 2025-04-15:
154f3ec0cfbfe0525a113edf6a039d1647ffe1ff

after a job is completed, the files are copied now to the media folder also.

that should finally solve it.

commit 12205bd8 (2025-04-15): improve naming of files to reflect a sequence, fix #500
closed 2025-04-15 by mgineer85