photobooth-app/photobooth-app #510: Can't use camera module 3 on app

Wihoui, 2025-04-25:
Hello,
i check tickets already created but i don't find my solution.
I use App on pi5 with BookWorm and i want to use a camera module 3.

I can see camera return when i do libcamera-hello.
But when i configure the App to use camera, i have an issue.

When i start App, i have the error message "Could not init/start backend (aquisition.py:85)
And when i want to take a picture, i have the same issue.
List index out of range (aquisition.py:84)

I test many things and i'm totaly lost...
Please can you help me ?

**System (please complete the following information):**
 - OS: Raspberry Pi Bookworm 64bit

mgineer85, 2025-04-25:
Yes, this is a bug in the latest version and fixed in the next.
Better disable the toggle next to the apply button in config, because the reloading fails for picamera currently.
Restart the whole app manually and you are fine.

mgineer85, 2025-04-25:
#505

Wihoui, 2025-04-26:
Thank you for this solution, i see app works well when i call it from user repo but not when i call it from cd ~/photobooth-data
i will test your solution

mgineer85, 2025-04-26:
Pls post logs and check the output onnthe terminal

mgineer85, 2025-04-27:
with the latest release, this bug is solved.   If you still have an issue, please open a post in the discussions sections including a log so we can see what is going on.

Have a great day ☀️

closed 2025-04-27 by mgineer85