photobooth-app/photobooth-app #511: [PiCamera2][7.1.0][RPi5] PiCamera2 error

Fre-d, 2025-05-01:
Hello,
I've had a problem since I upgraded from version 5.1.0 to version 7.1.0 (on RPi5). Everything was working fine: livestream and photo. But since the update, my camera (PiCamera2) no longer works, I get a ‘livestream error’ error, and the photos don't work either. Here are the error logs I have:
```
May 01 11:42:35 raspberrypi python[1053]: 2025-05-01 11:42:35,973 [    INFO] camera_config: {'use_case': 'video', 'transform': <libcamera.Transform 'identity'>, 'colour_space': <libcamera.ColorSpace 'Rec709'>, 'buffer_count': 3, 'queue': True, 'main': {'format': 'XBGR8888', 'size': (2304, 1296), 'stride': 9216, 'framesize': 11943936}, 'lores': {'format': 'YUV420', 'size': (1152, 648), 'stride': 1152, 'framesize': 1119744}, 'raw': {'format': 'RGGB_PISP_COMP1', 'size': (3840, 2160), 'stride': 3840, 'framesize': 8294400}, 'controls': {'NoiseReductionMode': <NoiseReductionModeEnum.Fast: 1>, 'FrameDurationLimits': (33333, 33333), 'FrameRate': 25}, 'sensor': {'bit_depth': 10, 'output_size': (3840, 2160)}, 'display': 'lores', 'encode': 'lores'} (picamera2.py:281)
May 01 11:42:35 raspberrypi python[1053]: 2025-05-01 11:42:35,974 [    INFO] camera_controls: {'FrameDurationLimits': (55543, 206049899, None), 'Brightness': (-1.0, 1.0, 0.0), 'AfRange': (0, 2, 0), 'LensPosition': (0.0, 32.0, 1.0), 'AfMetering': (0, 1, 0), 'AfWindows': ((0, 0, 0, 0), (65535, 65535, 65535, 65535), (0, 0, 0, 0)), 'AfPause': (0, 2, 0), 'Contrast': (0.0, 32.0, 1.0), 'ColourGains': (0.0, 32.0, None), 'AfTrigger': (0, 1, 0), 'AeFlickerPeriod': (100, 1000000, None), 'AwbMode': (0, 7, 0), 'AfMode': (0, 2, 0), 'HdrMode': (0, 4, 0), 'AwbEnable': (False, True, None), 'AfSpeed': (0, 1, 0), 'AeExposureMode': (0, 3, 0), 'CnnEnableInputTensor': (False, True, False), 'Saturation': (0.0, 32.0, 1.0), 'ExposureValue': (-8.0, 8.0, 0.0), 'AeConstraintMode': (0, 3, 0), 'ScalerCrop': ((408, 672, 145, 82), (408, 672, 3840, 2160), (408, 672, 3840, 2160)), 'AeFlickerMode': (0, 1, 0), 'Sharpness': (0.0, 16.0, 1.0), 'AnalogueGain': (1.0, 16.0, None), 'NoiseReductionMode': (0, 4, 0), 'StatsOutputEnable': (False, True, False), 'AeMeteringMode': (0, 3, 0), 'ExposureTime': (491, 206049113, None), 'AeEnable': (False, True, None), 'ScalerCrops': ((408, 672, 3840, 2160), (408, 672, 3840, 2160), (408, 672, 3840, 2160))} (picamera2.py:282)
May 01 11:42:35 raspberrypi python[1053]: 2025-05-01 11:42:35,974 [    INFO] controls: <Controls: {'NoiseReductionMode': <NoiseReductionModeEnum.Fast: 1>, 'FrameDurationLimits': (40000, 40000)}> (picamera2.py:283)
May 01 11:42:35 raspberrypi python[1053]: 2025-05-01 11:42:35,974 [    INFO] camera_properties: {'Model': 'imx519', 'UnitCellSize': (1220, 1220), 'ColorFilterArrangement': 0, 'Location': 2, 'Rotation': 0, 'PixelArraySize': (4656, 3496), 'PixelArrayActiveAreas': [(8, 48, 4656, 3496)], 'ScalerCropMaximum': (408, 672, 3840, 2160), 'SystemDevices': (20752, 20753, 20754, 20755, 20756, 20757, 20758, 20739, 20740, 20741, 20742), 'SensorSensitivity': 1.0} (picamera2.py:284)
May 01 11:42:35 raspberrypi python[1053]: 2025-05-01 11:42:35,974 [    INFO] stream quality Quality[self._config.videostream_quality]=<Quality.HIGH: 3> (picamera2.py:290)
May 01 11:42:35 raspberrypi python[1053]: 2025-05-01 11:42:35,975 [   ERROR] QSCALE (resilientservice.py:35)
May 01 11:42:35 raspberrypi python[1053]: Traceback (most recent call last):
May 01 11:42:35 raspberrypi python[1053]:   File "/home/pi/.local/pipx/venvs/photobooth-app/lib/python3.11/site-packages/photobooth/utils/resilientservice.py", line 43, in _run
May 01 11:42:35 raspberrypi python[1053]:     self.setup_resource()
May 01 11:42:35 raspberrypi python[1053]:   File "/home/pi/.local/pipx/venvs/photobooth-app/lib/python3.11/site-packages/photobooth/services/backends/picamera2.py", line 296, in setup_resource
May 01 11:42:35 raspberrypi python[1053]:     self._picamera2.start_encoder(self._mjpeg_encoder, FileOutput(self._lores_data), quality=Quality[self._config.videostream_quality])
May 01 11:42:35 raspberrypi python[1053]:   File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1746, in start_encoder
May 01 11:42:35 raspberrypi python[1053]:     _encoder.start(quality=quality)
May 01 11:42:35 raspberrypi python[1053]:   File "/usr/lib/python3/dist-packages/picamera2/encoders/encoder.py", line 231, in start
May 01 11:42:35 raspberrypi python[1053]:     self._start()
May 01 11:42:35 raspberrypi python[1053]:   File "/usr/lib/python3/dist-packages/picamera2/encoders/libav_mjpeg_encoder.py", line 62, in _start
May 01 11:42:35 raspberrypi python[1053]:     self._stream.codec_context.flags |= av.codec.context.Flags.QSCALE
May 01 11:42:35 raspberrypi python[1053]:                                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
May 01 11:42:35 raspberrypi python[1053]:   File "/usr/lib/python3.11/enum.py", line 789, in __getattr__
May 01 11:42:35 raspberrypi python[1053]:     raise AttributeError(name) from None
May 01 11:42:35 raspberrypi python[1053]: AttributeError: QSCALE
May 01 11:42:35 raspberrypi python[1053]: 2025-05-01 11:42:35,976 [CRITICAL] backend crashed, error: QSCALE (resilientservice.py:36)
May 01 11:42:35 raspberrypi python[1053]: 2025-05-01 11:42:35,976 [    INFO] trying to recover from crash (resilientservice.py:72)
```
I see an attribute in error (QSCALE) but I don't understand what it is, and I can't find any information on the subject.

Thanks for your help.

Fre-d, 2025-05-01:
It's Ok, i need to reinstall my camera : https://docs.arducam.com/Raspberry-Pi-Camera/Pivariety-Camera/Quick-Start-Guide/#use-libcamera-to-access-arducam-pivariety-camera

closed 2025-05-01 by Fre-d

mgineer85, 2025-05-01:
Yes, with arducam after each picamera/libcamera update you need to reinstall and hope arducam did not break it, unfortunately

juangabreil, 2025-05-24:
@Fre-d which distro have you installed on your rpi? I'm having a lot of problems with the AV library :(

mgineer85, 2025-05-24:
You need the latest 64 bit