photobooth-app/photobooth-app #512: Share Service / QR Download not working /reconnect loop

dtimes, 2025-05-02:
**Describe the bug**
QR sharing / upload service stopped working after update to latest version (server + client side).
same setup was working fine before, script is running on apache Webserver with default Debian config (server side). logs on Photobooth side look like a timeout/reconnect loop , not sure if this is blocking the actual upload or if it's another issue. Internet connection is stable and fast enough ( >200 mbit both directions) 

**To Reproduce**
Steps to reproduce the behavior:
1. scan arcade on Photo Booth
2. See error

**Expected behavior**
picture up and download working

**Log**
(qrshare.py:99)
2025-05-02 09:44:54,676 [    INFO] request timed out, error occured or shutdown requested (qrshare.py:164)
2025-05-02 09:44:54,676 [    INFO] restarting loop wait 1 second (qrshare.py:166)
2025-05-02 09:44:57,139 [    INFO] successfully connected to shareservice dl.php script (qrshare.py:74)

2025-05-02 09:45:05,148 [ WARNING] encountered shareservice connection issue. retrying. error: HTTPSConnectionPool(host='download.xxxx.de', port=443): Read timed out. (qrshare.py:99)
2025-05-02 09:45:05,149 [    INFO] request timed out, error occured or shutdown requested (qrshare.py:164)
2025-05-02 09:45:05,149 [    INFO] restarting loop wait 1 second (qrshare.py:166)
2025-05-02 09:45:08,669 [    INFO] successfully connected to shareservice dl.php script (qrshare.py:74)

2025-05-02 09:45:16,675 [ WARNING] encountered shareservice connection issue. retrying. error: HTTPSConnectionPool(host='download.xxxx.de, port=443): Read timed out. (qrshare.py:99)
2025-05-02 09:45:16,676 [    INFO] request timed out, error occured or shutdown requested (qrshare.py:164)
2025-05-02 09:45:16,676 [    INFO] restarting loop wait 1 second (qrshare.py:166)
2025-05-02 09:45:20,858 [    INFO] successfully connected to shareservice dl.php script (qrshare.py:74)
2025-05-02 09:45:21,050 [    INFO] getting latest logfile: log/photobooth_20250502.log (debug.py:21)

Php-error-log:

[02-May-2025 09:45:15 Europe/Berlin] Runtime error: photobooth did not upload the requested image within time :( no internet? service disabled?
[02-May-2025 09:45:15 Europe/Berlin] {"action":"download","id":"xxxxxxxxxxxxxxxxxxx"}
[02-May-2025 09:45:15 Europe/Berlin] []
[02-May-2025 09:45:15 Europe/Berlin] []

**System (please complete the following information):**
 - OS: debian 12 x64
 - Browser chrome / firefox 
 - Version latest release 

**Additional context**
apache error log is clean, not sure how to troubleshoot this, any help would be highly appreciated

mgineer85, 2025-05-02:
There have been changes made to the dl.php in one of the latest releases. Did you use the currentmost dl.php?

dtimes, 2025-05-02:
I skipped updating the dl.php on the server side for one or two releases (did not use the function and forgot about it) , just updated to the most recent dl.php (just adjusted the api key in the file).  I did some additional testing a few minutes ago and I think it's not caused by your adjustments. I've changed the url to point to another vserver (strangely with the exactly the same apache2 config) and I don´t have the timeout loop and the up/downloads are working.  If I sync the folders from server to server the download functionality works for both  , but if I change back the url in the Photo Booth-app the timeout is back again. not sure what´s causing the timeouts in the photobooth-> server connections, but it looks like it´s an issue with my setup and not your code. thanks again for your work

closed 2025-05-05 by mgineer85