photobooth-app/photobooth-app #520: timeout in standalone gallery returns to wrong main page

mgineer85, 2025-05-23:
hi, sorry for the late reply , I tried using the direct url from the documentation (link you provided) with the latest app version (second pc , starting chrome in kiosk mode pointing to the gallery only URL on the pc running photobooth-app). this works for some time / after start and the back button goes only back to the overview / desired gallery page. 

strangely if the system is idle for an extended time and it´s showing a single picture (not gallery thumbnails) the back button seems to change and if clicked the main photobooth screen is shown instead of the gallery (with photo taking action).  I did not try to troubleshoot / get an idea about the root cause yet but I rebooted the (gallery only) system and double checked the URL (which is fine ) and the issue stays the same.  the gallery/print only terminal changing into the photobooth view currently prevents me from using this feature

not sure if this is caused by the main photobooth also switching to idle mode / showing slideshow

_Originally posted by @dtimes in https://github.com/photobooth-app/photobooth-app/discussions/518#discussioncomment-13249957_

dtimes, 2025-05-25:
small update, even without reaching a timeout the second pc set to the view gallery only url switches to the countdown and stays in the "full photobooth mode" , this seems to be not related to the back button

dtimes, 2025-05-25:
did some testings, countdown start on display/gallery only pc starts if using buzzer / trigger via emulated keyboard ( https://photobooth-app.org/extras/buzzer/ ),  not sure if it´s random but I was not able to reproduce this with pressing the capture button on the touchscreen

closed 2025-05-27 by mgineer85

mgineer85, 2025-07-27:
Fix is included in v8.0.0rc1 now and ready to release if there are no major issues found in the rc very soon.
https://github.com/photobooth-app/photobooth-app/releases/tag/v8.0.0rc1