photobooth-app/photobooth-app #536: Counting the parameters (number of prints) in the print action

msmedien, 2025-06-10:
**Is your feature request related to a problem? Please describe.**
If I enable the parameter input in a print action and, for example, allow the user to make a maximum of three prints at once, the print action is still only counted once in the statistics overview. After an event, I then show (for example) 57 times the print action was executed in the statistics. However, 122 photos were actually printed. I have a DNP printer with 700 photos per roll, which makes tracking extremely difficult, and I constantly have to remove the printer and roughly estimate how many photos are left on the roll.

**Describe the solution you'd like**
A simple solution would be to count the value for the selected parameter number separately, for example, in parentheses after the value for the executed print action. This way, you can see how many times "Print" was pressed and how many photos were created as a result.

**Describe alternatives you've considered**
Alternatively, you could also list the value in a separate statistic. But that would eventually become too confusing if every share or print action received a separate statistic.

mgineer85, 2025-06-11:
Hi,
yeah so somewhere around here a multiplication needs to be added. 
Since it’s not clear which parameter in the config would affect the counter, the config needs to be extended so one can choose to enable a multiplication for each parameter.

dtimes, 2025-06-23:
does enabling the page log in cups do the job of tracking the consumption ? I think cups should see the right page count indepedent from the photobooth-app. would be nice to know because I´m planning to use this to track how many prints are left on a DNP printer (ordered, but not arrived yet)

mgineer85, 2025-06-25:
If you do not have a "number-of-copies" in the print, it can count it, yes.
I think using cups for this is actually a good idea. It seems you can enable such type of logging. Maybe it is displayed then in the cups backend or a simple script sums up the log output?

Jimmi123, 2025-08-16:
I activated the page log with
`PageLogFormat %{job-media-sheets-completed} %T`
in /etc/cups/cupsd.conf

This is what the log file /var/log/page_log looks like:
```
1 [16/Aug/2025:23:39:24 +0200]
2 [16/Aug/2025:23:41:14 +0200]
```
(No counting up here, each line is a print job.)

I can calculate the total using this command:
`awk '{ sum += $1 } END { print sum }' /var/log/cups/page_log`

Is there a way to write this value back to the front end using a script? Where is the counter stored?

@dtimes Would you share the webui?