photobooth-app/photobooth-app #540: Error processing the job

find-usdk, 2025-06-13:
**Describe the bug**
I’ve managed to get the webcam (Surface Pro 5) working, but when I press the Image (Take picture) button, it counts down just fine, but when it tries to save the picture, I get the error: 'Error processing the job. Please try again.'
However, if I press Boomerang instead, it works perfectly.

**To Reproduce**
Steps to reproduce the behavior:
1. Press Image (Take picture)
2. After countdown the error apears : 'Error processing the job. Please try again.'

**Expected behavior**
Processing and saving picture to gallery.

**Link**
[Discussions thread](https://github.com/photobooth-app/photobooth-app/discussions/537#discussioncomment-13446811)

**Screenshots**
[image]

**System (please complete the following information):**
 - OS: Windows 10 Pro 64bit
 - Browser Chrome Version 137.0.7151.69
 - Microsoft Surface Pro 5 Laptop
 - Front Webcam on Microsoft Surface Pro 5

**Additional context**
[mgineer85](https://github.com/mgineer85) think the internal cam just does not support jpeg streaming but only raw yuv format which is not expected in stills.

closed 2025-06-13 by mgineer85

mgineer85, 2025-06-18:
Confirmed it's working with normal mjpg cameras also, now.

There are all optional donation links in the github repo available :)

mgineer85, 2025-07-27:
Fix is included in v8.0.0rc1 now and ready to release if there are no major issues found in the rc very soon.
https://github.com/photobooth-app/photobooth-app/releases/tag/v8.0.0rc1

commit e37e7344 (2025-06-13): fix #540