photobooth-app/photobooth-app #553: Save boomerang / video as .gif instead of .mp4

PixelAlarm, 2025-07-16:
**Is your feature request related to a problem? Please describe.**
Since at least a few weeks (related to Nextcloud updates?), my .mp4 won't play in the browser when shared via my Nextcloud-DAV-Links on an iOS-Device. 

[image]

While investigating, this seems not to be an issue with the video itself, but with the html headers.

To achieve a workaround or more robust compatibility, it would be super cool to have the option to export as gif.

**Describe the solution you'd like**
Be able to select .gif as output video format.

**Describe alternatives you've considered**
For now I tried to use the commander to convert to gif after the event. This works pretty well, but this can't be linked to the custom QR code.

Nextcloud-WebDAV seems not to be editable to allow for inline preview.

Maybe other video formats could work instead.

mgineer85, 2025-07-16:
The nextcloud is not yet released. Maybe accidentally you disabled the video compatibility mode? iOS cannot replay then.

mgineer85, 2025-07-16:
Hmm, maybe it is more related to the nextcloud instance or php? 
Once the synchronizer is released it will be resolved , because there is a slim frontend to display the media files.
Does the playback work when you try to play the file from the box directly instead nextcloud?

PixelAlarm, 2025-07-18:
Thanks for your input.

Tested local access right now via "box.local:8000/media/full/{mediaitem_id}" and it is working as intended.
When trying to get the same file via Nextcloud, the media isn't played.

And I think this is because of the html header "content-disposition: attachment;" that Nextcloud delivery (Version 31)

Any .jpg as attachment will be displayed on iOS, any .gif can be displayed. But .mp4 will be marked broken / unplayable.
Bonus: The .gif can be the saved to the gallery directly via longpress.

So to be a bit more resilient, the gif-option would bring a big benefit in my opinion.

PixelAlarm, 2025-07-29:
I understand it is inefficient.
On my end the hardware would be sufficient, I guess :)

I’m using a cloud based Nextcloud Service. It‘s ready to use without administration and is easily integratable into the app. Now even easier with the synchronizer.
Without tinkering with the Nextcloud instance, mp4 will not be displayed on iOS from now on.

And to use the download page, I will have to get an additional hosting, I guess.

So having .gif as an option would allow for another sharing option.
I think this might be a niche request, so maybe I will look into the FTP solution.

mgineer85, 2025-08-06:
Hey, sorry, I will not add additional GIF conversion. 
I think opposite makes sense. Remove GIF completely and switch to mp4 also for the animations. 

A local solution could be to hook in the event when files were added to the db, write your own conversion and upload the files.

```python
@hookimpl
    def collection_files_added(self, files: list[Path]): ...
```