photobooth-app/photobooth-app #585: Wled not working anymore from 7.0.0

chris92q, 2025-08-13:
Hi there,

I was running on 4. somethIng for more than a year and had wled running including DMX etc for flash, amazing! 

I recently upgraded to 8.0.1 and everything was working, apart from the wled plugin. I set everything up as before, hat no errors in debug. For some reason the presets couldn’t be fired… the connection was clearly there. I then injected older versions, all the way down to 6.1.0. - all of a sudden it works perfektly fine again. 

What could I have done wrong? I checked the groups for dialout - that was all good.

mgineer85, 2025-08-14:
Which version did you came from?

Starting from v6 the WLED is now a plugin and not in the app core.  I usually add "breaking changes" to the release notes for every version when such a change takes place and you would need to update your config on your own.

So did you check the WLED tab in the config and it is setup properly?

If you dont get it working, please post the logs.

chris92q, 2025-08-14:
Hi there, I'll try and post the logs in a few days. 

I can from 4.0.0, then 4.1.5 - everything works perfectly fine there. 

Even in version 6.1.0 it works just fine - I adjusted the config and it imminently works, even though it's moved to the plugin - no problem. 

However, I can't get it working from 7.0.0 onwards. 
I configured everything - it just won't work. The serial port is definitely correct, at least it works in all the other versions. When I enter some random serial port, I get wled errors in the log. When I change it back to the correct serial port, there are no wled errors. 

I also noticed, that when I shut down the raspi, it takes forever to turn off - like it's stuck in some serial communication processes. When I turn off the wled plugin it reboots lightning fast again. 

I'll try and send the logs by the end of the week. 

Thanks for your super fast response :)! I love the Project!

mgineer85, 2025-08-15:
I really need the logs to see what is going on. can you share these, please?

mgineer85, 2025-08-15:
Also you use the provided settings for wled? Maybe there is a difference? Which wled version you use?

chris92q, 2025-08-17:
Right, I'm back - sorry i couldn't get working on it any earlier... 

Wled -> I was running version v. 0.14.4 (with DMX), I had no issues all the way up to 6.1.0. From 7.0.0 it stopt working, so I upgraded my wled to the newest version 0.15.1 (with DMX). It didn't solve the issue. I then downgraded photobooth to 4.1.5 again and everything was working just fine. So in the end the wled version made no difference. I also tried upgrading photobooth vs. installing on a clean system - doesn't make a difference. 

I have the logs attached now. I'm also attaching the logs from my v4.1.5 system, as this was/ is working fine. 

I changed absolutely nothing with my hardware switching between the versions - i only have one photobooth and a didn't touch the cableing etc. 

[photobooth_20250817 - booth v4.1.5 - wled v0.15.1 - all working.log](https://github.com/user-attachments/files/21824534/photobooth_20250817.-.booth.v4.1.5.-.wled.v0.15.1.-.all.working.log)

[photobooth_20250817 - booth v8.0.1 - wled v0.15.1 - wled not working.log](https://github.com/user-attachments/files/21824538/photobooth_20250817.-.booth.v8.0.1.-.wled.v0.15.1.-.wled.not.working.log)

I hope this helps :)!

chris92q, 2025-08-17:
Hey! Yes, I saw that too - there is no other instance accessing it - not that I know of. At least I didn't change anything and i guess that issue must have occurred in earlier versions as well. I don't have anything else running in that Raspi apart from the photobooth app. 
I checked the access - but that user (it's the only user) is all groups I found necessary, including the dialout group. 

I have a script running which disables and then enables usb 10 seconds after boot as I used to have issues with the camera getting recognised (that was version 4.0.0 back then). But even with a clean install and no such script i get the issue. 
That might explain why it started twice...

chris92q, 2025-08-22:
Hi there! 
I had problems with that too a while ago. However - I already have it removed....

When I enter the command it tells me that it's not installed and therefore couldn't be removed.

It's really frustrating. I've been trying and trying. I can't get it to work. The moment I roll back to 6.1.0 it works brilliantly!

chris92q, 2025-09-09:
Just another addition - I'm really a noob when it get to this but in terminal it shows the following: 

2025-09-09 00:48:14,470 [    INFO] WledService enabled and initialized, using port /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0 (wled.py:58)
2025-09-09 00:48:14,470 [   DEBUG] WLED (/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0)-resilient service starting (resilientservice.py:94)

and then, a little later it gives us the services status: 

2025-09-09 00:48:14,825 [    INFO] services status: ['LoggingService: started', 'PluginManagerService: started', 'AquisitionService: started', 'MediacollectionService: started', 'InformationService: started', 'ProcessingService: started', 'SystemService: started', 'ShareService: started', 'GpioService: disabled', 'QrShareService: disabled', 'FileTransferService: disabled', 'ConfigurationService: started'] (container.py:63)

If WLED is a service, why isn't it showing in the services status?

mgineer85, 2025-09-10:
Can you also describe what the device is? ESP32? What board?    And you hopefully got rid of the script to enable/disable the USB/serial ports?

chris92q, 2025-09-10:
I will try the approach you suggest as soon as I get back home. I'll also check which board i'm using - it's definitly an ESP32 with USB-C and not a D1 Mini - I'm just not sure which model (WROOM or C3). 
I can confirm that I got rid of my "USB-enable/disable-script", as It's a complete fresh install and I didn't need it anymore for 6.1.0 either...

mgineer85, 2025-09-10:
```

Pin usage limitation
If GPIO3 is allocated (e.g. for LED output), all Serial functionality except debug output is unavailable.
If GPIO1 is allocated, all Serial output is disabled, including the JSON API response, Improv, and tpm2 output.

```

From wled docs / which pins you use for your leds?

chris92q, 2025-09-10:
> Hi, for any reason it seems the setup of WLED stalls. Maybe this was just ignored in older versions, not 100% sure about this, yet and I cannot see why it would stall.
> 
> ```
> import json
> import logging
> import time
> 
> import serial
> 
> logger = logging.getLogger(__name__)
> 
> PORT = "/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0"
> 
> _serial = serial.Serial(
>     port=PORT,
>     baudrate=115200,
>     bytesize=8,
>     timeout=1,
>     write_timeout=1,
>     stopbits=serial.STOPBITS_ONE,
>     rtscts=False,
>     dsrdtr=False,
> )
> 
> print(_serial.is_open)
> 
> print(_serial.write(b"v"))
> 
> 
> # readline blocks for timeout seconds (set to 1sec on init), afterwards fails
> wled_response = _serial.readline()
> print(wled_response)
> 
> wled_detected = "WLED" in wled_response.decode("ascii").strip()
> print(wled_detected)
> 
> print("set to countdown")
> _serial.write(json.dumps({"ps": 2}).encode())
> time.sleep(2)
> 
> print("set to standby")
> _serial.write(json.dumps({"ps": 1}).encode())
> 
> 
> _serial.close()
> ```
> 
> You should be able to save it to a file like `wled_script.py` and then execute it as `python wled_script.py`.
> 
> Hopefully we see where the script hangs and if an error it thrown with this very simple approach. While you execute, it should set the WLED ringlight to countdown preset for 2 seconds and then go to standby again.

This works perfectly! 

photobooth@photobooth:~ $ python /home/photobooth/Desktop/wled_script.py
True
1
b'WLED 2508020\r\n'
True
set to countdown
set to standby
photobooth@photobooth:~ $

chris92q, 2025-09-10:
YES - OMG!!!! I think you got it - I am running the LED Ring Data on GPIO 16, but the DMX goes out to the MAX485 on GPIO1.
I removed the second output (DMX, GPIO1) and only let the LED-Ring on GPIO16 in the WLED LED-Setup - then it works with the photobooth app! :D 
You might just have ended my nightmares... I'll  now try and rewire ESP32 -> MAX485 with some other GPIO and see if it works.
I can't believe it. But so strange that it works with all versions before 6.1.0. 
I'll let you know how it went after changing the GPIOs... 
THANK YOU!