photobooth-app/photobooth-app #599: Print Button not visible after picture

nifroidl, 2025-09-14:
**Describe the bug**
The "print" Button is not visible after the countdown and the pictore was made. 

BUT: When I activate "Gallery Show Shareprint", the print Button is visible in the Gallery AND after the picture was taken. I only want the button there when the picture is taken and NOT in the Gallery. I thought this was possible. I don't know what happend here...

**To Reproduce**
Steps to reproduce the behavior:
1. Click on 'Picture'
4. See error

**Expected behavior**
After the picture was made, the print Button should be visible for the time the picture is shown. In the Gallery, the print button should not be visible.

**System (please complete the following information):**
 - OS: Raspberry Pi Bookworm 64bit
 - Browser chrome and safari
 - Version 5.1.1

**Additional context**
Add any other context about the problem here.

mgineer85, 2025-09-14:
Hi, there is no actual distinction between the itempresenter and the detail view in the gallery. 
So it works as implemented correctly.

We could consider separate settings as enhancement for the future.

nifroidl, 2025-09-14:
Are you sure? I am pretty sure this feature already worked in the past. Just for my understanding: For what feature is the Button "Gallery Show Shaerprint" ?