photobooth-app/photobooth-app #602: Commander plugin does not trigger on video recording events

linuxmaster85, 2025-09-15:
Hello,
I am using Photobooth with the Commander plugin to control a Shelly 1 (relay for my ring light).
The idea:
Turn light ON when a video recording starts
Turn light OFF when recording stops
I created Commander actions with events like record, start, stop, and finished.
Unfortunately, none of these seem to fire during video recording. The URLs are never called.
Debug log shows Commander is loaded, but during a video recording no relevant events appear (only init/start/stop for photos, not for videos).
Here is an excerpt from my log:

[Commander] registered
... (no record_start/record_stop or similar events when recording video) ...

[2025-09-15 21:08:01] [Commander] Registered
[2025-09-15 21:08:12] Starting video recording (ffmpeg)
[2025-09-15 21:08:23] Video recording finished
# -> No Commander hook fired here

Questions:
Are there currently Commander hooks for video recording (record_start / record_stop)?
If not, could you please add hooks for video begin/end, so external devices (lights etc.) can be triggered?
Or is there another recommended way to trigger HTTP requests specifically on video start/stop?
This would allow integrations like Shelly/WLED to react to video recording the same way they do for photos.
Thanks a lot!

mgineer85, 2025-09-15:
Via wled plugin you can use presets to set different light scenarios.

The commander plugin should also fire an event. Did you try if the capture hook fires even for the record?

mgineer85, 2025-09-16:
Regarding the bind error and the device being blocked, please ensure you check
- https://photobooth-app.org/help/troubleshooting/#bind-error-during-app-startup
- https://photobooth-app.org/help/faq/#gphoto2-camera-found-but-no-access

This should be independent from the commander, but we will see.

For debugging purposes it might also be easier to just use the virtual camera and enable hardware later again.

To find out how many instances of the app run, you could check
```sh
ps ax | grep python
```

From the log I cannot see that you configured an event for "capture_video" but only for "capture". The first would trigger only for video captures, the second for all captures.
Also there is no command configured for "captured" which should turn the light off.

```
...
2025-09-16 18:03:32,184 [    INFO] 1 tasks to run for capture: [TaskHttpRequest(enabled=True, name='Ringlicht an Video', event=['capture'], wait_until_completed=True, timeout=5, url=HttpUrl('http://192.168.0.12/relay/0?turn=on'), method='get', body_parameters_as_json=True, parameter=[])] (commander.py:166)
2025-09-16 18:03:32,184 [    INFO] run http request 'http://192.168.0.12/relay/0?turn=on' (commander.py:62)
2025-09-16 18:03:32,326 [   ERROR] error sending http request, error None (commander.py:85)
2025-09-16 18:03:32,327 [    INFO] 0 tasks to run for capture_video: [] (commander.py:166)
...
2025-09-16 18:03:37,333 [    INFO] 0 tasks to run for captured: [] (commander.py:166)
...
```

I have no hardware to test here, so I am just guessing why there is the http error:
Try to set only the http://192.168.0.12/relay/0 in the URL field. Use the parameter field to set "turn" = "on"

Probably like this to turn the video light on

[image]

And a second entry (use ADD to create a new item in the config) to turn the light of when CAPTURED:

[image]

linuxmaster85, 2025-09-16:
Thank you very much, Michel 🙏
I really want to thank you for your quick support and for adding the new capture_video event so fast. With your explanation and the adjustments, my setup is now working perfectly – the ring light reliably switches on and off with video recording.
It’s great to see how actively and thoughtfully you support the project. Your help made a big difference for me, and I really appreciate the time and effort you put into this.
Thanks again for your support and for keeping this project moving forward! 🙌

mgineer85, 2025-09-17:
Thank you for your kind words :)

I added some initial description for the plugin to the docs which should be helpful for future users: https://photobooth-app.org/setup/configuration/commander/

Closing this one now, if you have issues, feel free to post again here or in separate issues.

closed 2025-09-17 by mgineer85

linuxmaster85, 2025-09-17:
Hi, I am using the Commander plugin to switch a ring light via Shelly when a video recording starts. At the moment the light only turns on exactly at the same time the recording begins, which with my Canon EOS 77D leads to the first 1–2 seconds of the video being overexposed until the camera’s auto-exposure has settled. It would be very helpful if there was a way to trigger the light a little earlier, for example by adding a capture_video_preroll event, allowing an offset configuration for Commander actions (e.g. run X ms before/after the event), or by letting the video countdown also fire the Commander hook. This way the light could already be on 1–2 seconds before recording, giving the camera time to adjust and avoiding the overexposed start. If such a possibility already exists I would be very grateful for a hint. I really value how responsive and supportive the project has been so far, it makes a big difference for setups like mine. Thanks a lot for considering this idea!

linuxmaster85, 2025-09-22:
I tested again: for photos I use a dedicated flash that is built into the box, so using the regular countdown event would not work — otherwise I’d have “holes” in my setup since the ring light would turn on for photos as well.
During collage mode I also noticed that sometimes the ring light gets turned off after the second picture, which makes the behavior a bit inconsistent.
If at some point in the future a dedicated capture_video_preroll event (maybe with a configurable second-offset or duration) could be added, that would be really great and solve the issue perfectly for video.
I also completely understand that this is more of a special request, and if it’s too much effort that’s fine.
Still, the project is amazing — thanks a lot for all the work you put into it! 🙏

mgineer85, 2025-09-24:
I added what I described before to a new branch. It seems to work well, but I did not test it too much yet, also some automated tests need to be added.

Are you able to test a separate branch on an experimental setup?

linuxmaster85, 2025-09-28:
Thanks a lot for the new branch — I’m happy to test it!
Quick question: what’s the recommended way to install it with pipx? Just uninstall and then run something like

pipx install git+https://github.com/andi34/photobooth-app.git@branchname

with your branch name, or do you suggest another way?
And should I mainly check the new filtered/delayed events, or also test photos, collage, gif, video to catch possible side effects?
Really appreciate your work on this

mgineer85, 2025-09-29:
Hey, you are right, here is some documentation about it: https://photobooth-app.org/setup/update/#update-to-development-versions

So I would uninstall it, create a backup of the data-dir (just to be safe, I don't think it would happen anything) and install from the branch.

Later you do the same to get back to the regular version.

It would be great if you can test multiple configurations, to ensure it works as intended.   I did only quick tests and I *think* it works correctly, but I don't use the commander actually.

linuxmaster85, 2025-09-29:
Hi Michael,
thanks a lot for providing the commander-plugin-filter-jobtypes branch — I was able to install and run it successfully.
I just did the first tests with my setup:
The filter options and delay settings now show up in the Commander config, so the branch is definitely active.
However, the behavior is not yet as expected: the ring light turns on after the capture/video has finished, but it does not turn off again afterwards.
I’ll attach a screenshot of my Commander configuration so you can see how I set it up. Maybe I misunderstood how to combine the filter + delay, or maybe the trigger order is still slightly off.
In any case: it’s already a huge step forward to have these filter and delay options available — thank you for your quick work on this feature

[image]

https://github.com/user-attachments/assets/1c1573d4-cb47-446e-86e3-dd2d581b2756

mgineer85, 2025-09-29:
Ja die config ist falsch. Es muss das Countdown Event mit Filter Video für Licht an genutzt werden. Am besten ohne Delay, denke ich.

mgineer85, 2025-09-29:
Capture video triggers erst wenn die Aufnahme startet und dann 5s Delay

linuxmaster85, 2025-09-30:
Genau diese „Synapsenbildung“ hat mir noch gefehlt – jetzt läuft alles perfekt

Ich habe folgendes getestet:
Video mit Ringlicht
Foto
Collage
GIF
Drucken
Drucksperre bei Video und GIF funktioniert ebenfalls.
Anbei noch ein Screenshot der Einstellungen sowie ein kurzes Video, wie es in der Praxis aussieht.

Nochmals 1000 Dank – macht richtig Spaß mit der Photobooth!

[image]
[image]

https://github.com/user-attachments/assets/d9eed9ff-4a9b-4093-87fd-45da6ce835b7

mgineer85, 2025-10-05:
Nicht nötig, ist alles geblieben wie es ist. Kannst einfach auf 8.1 aktualisieren, wenn die finale version raus ist.