photobooth-app/photobooth-app #603: Prevent Print hotkey from triggering on video results

linuxmaster85, 2025-09-16:
Hello,  

I am using Photobooth with 4 arcade buttons mapped to different functions:  
- Green = Photo (`e`)  
- Yellow = Collage (`s`)  
- Red = GIF (`y`)  
- Blue = Video (`p`)  
- White = Print (`w`)  

Everything works fine, except one issue:  
When a **video is recorded** and the final video is displayed, if someone presses the **Print** button (`w`), Photobooth tries to send the video file (MP4) to the printer.  
My Canon Selphy CP1500 then goes into an error state because it cannot print videos.  

**Question:**  
Is there a way to disable or hide the print function when a video (or GIF) is being displayed?  
Ideally, the `w` hotkey for printing should only be active for photo/collage results, not for video/GIF results.  

This would prevent the printer from receiving unsupported files and throwing errors.  

Thanks a lot for your help and for this amazing project! 🙏

[image]
[image]

mgineer85, 2025-09-17:
8.0.6 released, fixes it, so I close here.

linuxmaster85, 2025-09-17:
A huge thank you 🙏  

I’m honestly amazed by how quickly you responded and even released a dedicated new version (8.0.6) just for this issue – and it works perfectly! 🎉  

It’s rare to see such fast and thoughtful support in open source projects.  
Really appreciate the time and effort you put into making Photobooth better with every release.  

This means a lot to me and makes my setup so much more reliable.  
Thanks again for this outstanding support – you are doing an awesome job! 🙌